Mange/rtl8192eu-linux-driver #274: Not working for kernel headers 5.17.9

AndrGutierrez, 2022-05-22:
I upgraded my arch linux and it says 

`missing 5.17.9-arch1-1 kernel modules tree for module rtl8192eu/1.0`

My system does not recognize my wifi adapter anymore

CGarces, 2022-05-22:
The code should wok on 5.17.x since #263

I assume that you are using some AUR package like https://aur.archlinux.org/packages/rtl8192eu, you are not building the module as instructed in the [Reamde.md](https://github.com/Mange/rtl8192eu-linux-driver#building-and-installing-using-dkms)

You should send your questions to the comments section on the AUR package

closed 2022-05-22 by CGarces

AndrGutierrez, 2022-05-22:
No, I cloned this repo and installed it as it said in the readme, also when I run dkms install says that it is missing linux-headers 5.16.11