Mange/rtl8192eu-linux-driver #275: kernel error and system crash with hostapd: rtw_mlme_ext.c (issue_beacon, usb_write8, Debian-11)

kolAflash, 2022-05-27:
I'm getting kernel errors and even complete system crashes when using with hostapd on Debian-11.

At least one client has to be connected and I can transfer data for a moment. But after less than 2 minutes I get erors in dmesg and the client disconnects and won't reconnect anymore.
See: [kern-log.txt](https://github.com/Mange/rtl8192eu-linux-driver/files/8789568/kern-log.txt)
Kernel: Debian-11 kernel 5.10.0-14-amd64 (a.k.a. upstream 5.10.113-1)

Stopping hostapd and doing `rmmod 8192eu; modprobe 8192eu` sometimes fixes the problem. But it quickly reappears.
And sometimes `rmmod 8192eu` also crashes the whole system! (no response, so unfortunately I can't provide logs)

&nbsp;

&nbsp;

### Details

**Device**
Vendor: TP-Link
Name: 300Mbps High Gain Wireless USB-Adapter
Model:TL-WN822N(EU) Ver:5.0

I only experienced this problem with this client:
[Fairphone 3+](https://en.wikipedia.org/wiki/Fairphone_3#Fairphone_3+) running [/e/OS 0.21-q-20220121158536-dev-FP3](https://en.wikipedia.org/wiki//e/_(operating_system) / Android-10).

My notebook ([HP EliteBook 735 G6](https://web.archive.org/web/20191213065546/https://support.hp.com/us-en/document/c06386926), rtw88_8822be wireless) worked fine for more than 2 minutes and transferred with up to 2 MByte/s. But I didn't do extensive testing.
Also I didn't test any other client devices.

hostapd.conf:
```
interface=wlan0
bridge=br0
ssid=XXX
ieee80211d=1
country_code=DE
hw_mode=g
channel=13
auth_algs=1
wpa=2
rsn_pairwise=CCMP
wpa_key_mgmt=WPA-PSK
wpa_passphrase=XXX
ieee80211n=1
```

CGarces, 2022-06-03:
Duplicate of #273

Maybe you can try to build from 1401b54fd04fb46a36d9e3e7b3f7dff0194cc00d to ensure that I haven't break something after the code cleanup.

kryshen, 2022-06-15:
I'm getting same errors with the current version of the driver (6ba1f320963376f15ea216238c0b62ff3e71fa82).
TL-WN822N Ver. 4, hostapd on Debian 11, kernel 5.10.0-15-amd64.

Build from 1401b54fd04fb46a36d9e3e7b3f7dff0194cc00d appears to work without issues.

CGarces, 2022-06-15:
Ouch, so if you confirm that https://github.com/Mange/rtl8192eu-linux-driver/commit/1401b54fd04fb46a36d9e3e7b3f7dff0194cc00d works fine, I have broke something.
Unfortunately I don't have a device to made a git bisect and identify the offending commit

CGarces, 2022-06-15:
@jschultz can you test a build with https://github.com/Mange/rtl8192eu-linux-driver/commit/1401b54fd04fb46a36d9e3e7b3f7dff0194cc00d ?

jschultz, 2022-06-24:
Sorry I can't even replicate the kernel panic. But I did try to use hostapd with [1401b54](https://github.com/Mange/rtl8192eu-linux-driver/commit/1401b54fd04fb46a36d9e3e7b3f7dff0194cc00d) and it showed the same behaviour as the build I have been using. Which is that when the first client tries to connect, the connection is disconnected, and henceforth the signal appears to clients to have WEP encryption.

dukzcry, 2022-10-18:
I have the same issue. Done bisect

3555bdb25800fe9fb280c64663ab2e3a46762d63 `is the first bad commit`
e73d0db27b33e85881aef1f1e1f33fa715e103be and 41fddb43b3a351fce500cdc867807bfa2d3151c3 are last working

git bisect log
```
git bisect start
# bad: [e0f967cea1d0037c730246c572f7fef000865ff7] Merge pull request #265 from CGarces/kernel_517
git bisect bad e0f967cea1d0037c730246c572f7fef000865ff7
# good: [1401b54fd04fb46a36d9e3e7b3f7dff0194cc00d] Remove empty header file
git bisect good 1401b54fd04fb46a36d9e3e7b3f7dff0194cc00d
# good: [2f0a5e695bcdbc4088cbc0b561345bd23193f3ce] remove ethernet.h header file
git bisect good 2f0a5e695bcdbc4088cbc0b561345bd23193f3ce
# bad: [51d0819382215a83a78adcdfe5cf87c857e602aa] Use cpu_to_le16() insted of RTW_PUT_LE16()
git bisect bad 51d0819382215a83a78adcdfe5cf87c857e602aa
# good: [355a2df5ab7d103c69ed18a40d872340f5ad7bb6] Remove function rtw_end_of_queue_search()
git bisect good 355a2df5ab7d103c69ed18a40d872340f5ad7bb6
# bad: [2cd85b67990872c9d98f330cdf90a79f891cee6e] remove unused rtw_ieee80211_bar structure definition
git bisect bad 2cd85b67990872c9d98f330cdf90a79f891cee6e
# bad: [c5f30cb5ccc589309463945ecb129f729a6a36fd] remove enum WIFI_FRAME_SUBTYPE
git bisect bad c5f30cb5ccc589309463945ecb129f729a6a36fd
# bad: [3555bdb25800fe9fb280c64663ab2e3a46762d63] Replace custom definitions with existing ones at linux/ieee80211.h.
git bisect bad 3555bdb25800fe9fb280c64663ab2e3a46762d63
# good: [e73d0db27b33e85881aef1f1e1f33fa715e103be] Remove empty rtw_mfree_xmit_priv_lock() Remove rtw_mfree_xmit_priv_lock() because it is empty. Remove its only call from within _rtw_free_xmit_priv().
git bisect good e73d0db27b33e85881aef1f1e1f33fa715e103be
# first bad commit: [3555bdb25800fe9fb280c64663ab2e3a46762d63] Replace custom definitions with existing ones at linux/ieee80211.h.
```

CGarces, 2022-10-30:
@dukzcry 
Thanks for your help with the bisect
Can you please test #291?

dukzcry, 2022-10-31:
@CGarces Thank you! Testing results:
There is no more kernel crash, but the problem still persists that Android client disassociates after few minutes and also looses network settings like password. There is no such issue with https://github.com/Mange/rtl8192eu-linux-driver/commit/41fddb43b3a351fce500cdc867807bfa2d3151c3 revision
```
Oct 31 10:11:15 robocat hostapd[479010]: legacy: STA 08:e9:f6:0e:93:06 IEEE 802.11: associated
Oct 31 10:11:15 robocat hostapd[479010]: legacy: AP-STA-CONNECTED 08:e9:f6:0e:93:06
Oct 31 10:11:15 robocat hostapd[479010]: legacy: STA 08:e9:f6:0e:93:06 RADIUS: starting accounting session 28A4ACFDC4EE4157
Oct 31 10:11:15 robocat hostapd[479010]: legacy: STA 08:e9:f6:0e:93:06 WPA: pairwise key handshake completed (RSN)
Oct 31 10:11:15 robocat hostapd[479010]: legacy: EAPOL-4WAY-HS-COMPLETED 08:e9:f6:0e:93:06
Oct 31 10:11:15 robocat hostapd[479010]: legacy: STA 08:e9:f6:0e:93:06 RADIUS: starting accounting session 28A4ACFDC4EE4157
Oct 31 10:11:15 robocat hostapd[479010]: legacy: STA 08:e9:f6:0e:93:06 WPA: pairwise key handshake completed (RSN)
Oct 31 10:13:52 robocat hostapd[479010]: legacy: STA 08:e9:f6:0e:93:06 IEEE 802.11: disassociated
Oct 31 10:13:52 robocat hostapd[479010]: legacy: AP-STA-DISCONNECTED 08:e9:f6:0e:93:06
Oct 31 10:13:52 robocat hostapd[479010]: legacy: STA 08:e9:f6:0e:93:06 IEEE 802.11: disassociated
```

CGarces, 2022-10-31:
Hi!

I don't have a rtl8192eu device to test.
Maybe the disassociated client is caused by another commit.
I'll be hard to bisect again unless I rebase #291 into https://github.com/Mange/rtl8192eu-linux-driver/commit/3555bdb25800fe9fb280c64663ab2e3a46762d63, create an specific branch for testing and someone with the hardware bisect again.

CGarces, 2022-10-31:
@dukzcry 
Please try to use this branch for bisect 
https://github.com/CGarces/rtl8192eu-linux-driver/tree/disassociated_fix

dukzcry, 2022-10-31:
@CGarces Results:
80ef43ca45fcf3c39dd6c6af0cd2374787a130a6 has reproducable disassociation issue
https://github.com/CGarces/rtl8192eu-linux-driver/commit/fd5795b1cf9603336ef7b9bcae37d3b63af0b765 works fine

git bisect didn't find anything beyond 3555bdb25800fe9fb280c64663ab2e3a46762d63

CGarces, 2022-10-31:
LoL

If https://github.com/CGarces/rtl8192eu-linux-driver/commit/fd5795b1cf9603336ef7b9bcae37d3b63af0b765 works fine in theory should work fine https://github.com/Mange/rtl8192eu-linux-driver/pull/291/commits/80ef43ca45fcf3c39dd6c6af0cd2374787a130a6 
¿do I something wrong in the PR #291 and fixed in the [rebase](https://github.com/CGarces/rtl8192eu-linux-driver/tree/disassociated_fix) of just by a lucky mistake?

I'll review #291 tomorrow.

CGarces, 2022-10-31:
Opps, yes the PR is incomplete and I need to review everything again
At least one line is incorrect.
https://github.com/CGarces/rtl8192eu-linux-driver/blob/update_beacon_fix/core/rtw_ap.c#L1820

CGarces, 2022-10-31:
Ok, was easier than excepted.
I just forgot to push one file, but was on my local machine #291 should works fine now.

dukzcry, 2022-10-31:
This works fine now, thanks!

closed 2022-10-31 by CGarces