Mange/rtl8192eu-linux-driver #278: New Kernel update not showing wifi

nb5691, 2022-07-13:
I have Ubuntu 20.04.4 LTS desktop with kernel 5.13.0-52-generic and my wifi device DWA 131 with rtl8192eu driver was
functioning perfectly after every software update till this kernel version. Today after software update with kernel 5.15.0-41-generic, my wifi is not getting detected. Kindly help and pls let me know the fix, procedure for the same. Thanks and Regards.

nb5691, 2022-07-13:
> How have you installed the driver?
@CGarces: I have used the following link and steps mentioned in the same page, installed the driver and the same was working great till the recent kernel update today. Link is as follows :

https://github.com/Mange/rtl8192eu-linux-driver

CGarces, 2022-07-13:
So I assume that you are using DKMS.
You have any error log?
Have you tried to install it again?

nb5691, 2022-07-13:
Yes, I am using DKMS as mentioned in the steps. After kernel update, I don't have any error and error log file, simply wifi is not showing up. Hence I am reverting to older kernel 5.13.0-52-generic using Ubuntu GRUB menu and wifi works and able to access internet.