Mange/rtl8192eu-linux-driver #284: New Kernel (5.19.2) update  DKMS build failed

dafengsu7, 2022-08-18:
``` shell
DKMS make.log for 8192eu-r269.6ba1f32 for kernel 5.19.2-arch1-1 (x86_64)
Fri Aug 19 01:19:14 AM CST 2022
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.19.2-arch1-1/build M=/var/lib/dkms/8192eu/r269.6ba1f32/build  modules
make[1]: Entering directory '/usr/lib/modules/5.19.2-arch1-1/build'
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_mlme.o
In file included from ./include/linux/string.h:253,
                 from ./include/linux/bitmap.h:11,
                 from ./include/linux/cpumask.h:12,
                 from ./arch/x86/include/asm/cpumask.h:5,
                 from ./arch/x86/include/asm/msr.h:11,
                 from ./arch/x86/include/asm/processor.h:22,
                 from ./arch/x86/include/asm/timex.h:5,
                 from ./include/linux/timex.h:67,
                 from ./include/linux/time32.h:13,
                 from ./include/linux/time.h:60,
                 from ./include/linux/stat.h:19,
                 from ./include/linux/module.h:13,
                 from /var/lib/dkms/8192eu/r269.6ba1f32/build/include/basic_types.h:37,
                 from /var/lib/dkms/8192eu/r269.6ba1f32/build/include/drv_types.h:26,
                 from /var/lib/dkms/8192eu/r269.6ba1f32/build/include/../hal/phydm/phydm_types.h:215,
                 from /var/lib/dkms/8192eu/r269.6ba1f32/build/include/../hal/phydm/phydm_precomp.h:29,
                 from /var/lib/dkms/8192eu/r269.6ba1f32/build/include/hal_data.h:20,
                 from /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_mlme.c:17:
In function ‘fortify_memcpy_chk’,
    inlined from ‘rtw_restructure_ht_ie’ at /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_mlme.c:4744:2:
./include/linux/fortify-string.h:344:25: warning: call to ‘__write_overflow_field’ declared with attribute warning: detected write beyond size of field (1st parameter); maybe use struct_group()? [-Wattribute-warning]
  344 |                         __write_overflow_field(p_size_field, size);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_ap.o
/var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_ap.c: In function ‘rtw_ap_update_chbw_by_ifbmp’:
/var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_ap.c:4432:42: warning: the comparison will always evaluate as ‘true’ for the address of ‘padapters’ will never be NULL [-Waddress]
 4432 |                 if (!(ifbmp & BIT(i)) || !dvobj->padapters)
      |                                          ^
In file included from /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_ap.c:17:
/var/lib/dkms/8192eu/r269.6ba1f32/build/include/drv_types.h:1017:19: note: ‘padapters’ declared here
 1017 |         _adapter *padapters[CONFIG_IFACE_NUMBER];/*IFACE_ID_MAX*/
      |                   ^~~~~~~~~
/var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_ap.c:4452:42: warning: the comparison will always evaluate as ‘true’ for the address of ‘padapters’ will never be NULL [-Waddress]
 4452 |                 if (!(ifbmp & BIT(i)) || !dvobj->padapters)
      |                                          ^
/var/lib/dkms/8192eu/r269.6ba1f32/build/include/drv_types.h:1017:19: note: ‘padapters’ declared here
 1017 |         _adapter *padapters[CONFIG_IFACE_NUMBER];/*IFACE_ID_MAX*/
      |                   ^~~~~~~~~
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/8192eu/r269.6ba1f32/build/core/rtw_btcoex_wifionly.o
[... 32 lines cut ...]
      |            ^~~~
/var/lib/dkms/8192eu/r269.6ba1f32/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_external_auth_status’:
/var/lib/dkms/8192eu/r269.6ba1f32/build/os_dep/linux/ioctl_cfg80211.c:9786:35: warning: the comparison will always evaluate as ‘true’ for the address of ‘pmkid’ will never be NULL [-Waddress]
 9786 |                 if (params->pmkid != NULL) {
      |                                   ^~
In file included from /var/lib/dkms/8192eu/r269.6ba1f32/build/include/osdep_intf.h:111,
                 from /var/lib/dkms/8192eu/r269.6ba1f32/build/include/drv_types.h:83:
/var/lib/dkms/8192eu/r269.6ba1f32/build/include/../os_dep/linux/ioctl_cfg80211.h:207:12: note: ‘pmkid’ declared here
  207 |         u8 pmkid[PMKID_LEN];
      |            ^~~~~
/var/lib/dkms/8192eu/r269.6ba1f32/build/os_dep/linux/ioctl_cfg80211.c: At top level:
/var/lib/dkms/8192eu/r269.6ba1f32/build/os_dep/linux/ioctl_cfg80211.c:9866:20: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, unsigned int)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct net_device *)’ [-Werror=incompatible-pointer-types]
 9866 |         .stop_ap = cfg80211_rtw_stop_ap,
      |                    ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/8192eu/r269.6ba1f32/build/os_dep/linux/ioctl_cfg80211.c:9866:20: note: (near initialization for ‘rtw_cfg80211_ops.stop_ap’)
/var/lib/dkms/8192eu/r269.6ba1f32/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_wdev_unregister’:
/var/lib/dkms/8192eu/r269.6ba1f32/build/os_dep/linux/ioctl_cfg80211.c:10135:17: error: ‘struct wireless_dev’ has no member named ‘current_bss’
10135 |         if (wdev->current_bss) {
      |                 ^~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:249: /var/lib/dkms/8192eu/r269.6ba1f32/build/os_dep/linux/ioctl_cfg80211.o] Error 1
make[1]: *** [Makefile:1849: /var/lib/dkms/8192eu/r269.6ba1f32/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.19.2-arch1-1/build'
make: *** [Makefile:2287: modules] Error 2
````

CGarces, 2022-08-18:
working on that, but maybe you need to wait until weekend.
Feel free to send a PR adapting this:
https://github.com/aircrack-ng/rtl8812au/pull/992

dafengsu7, 2022-08-19:
> working on that, but maybe you need to wait until weekend. Feel free to send a PR adapting this: [aircrack-ng/rtl8812au#992](https://github.com/aircrack-ng/rtl8812au/pull/992)

get it, thank you !

CGarces, 2022-08-21:
Fixed by ec5c2446db5f3380d7468b1e71fe9db67cbc8a8b

closed 2022-08-21 by CGarces