Mange/rtl8192eu-linux-driver #289: Can't initialise driver

robcalcroft, 2022-10-18:
When i run `ifconfig` i see the interface `wlx2887ba3aaf01 ` showing up but when i go to start wpa_supplicant i get this error:

```
me@server:~$ wpa_supplicant -B -i wlx2887ba3aaf01 -c /etc/wpa_supplicant/wpa_supplicant.conf 
Successfully initialized wpa_supplicant
nl80211: deinit ifname=wlx2887ba3aaf01 disabled_11b_rates=0
wlx2887ba3aaf01: Failed to initialize driver interface
```

```
$ cat /proc/version
Linux version 4.15.0-194-generic (buildd@lcy02-amd64-052) (gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04)) #205-Ubuntu SMP Fri Sep 16 19:49:27 UTC 2022
```

Do i need to compile differently for a amd64 system?

CGarces, 2022-10-18:
amd64  should use the standard CONFIG_PLATFORM_I386_PC  = y

Please compile the kernel from
https://github.com/clnhub/rtl8192eu-linux/tree/5.11.2.1
And ping me with the results to understand if the problem is caused by my custom modifications.

CGarces, 2022-10-18:
I don't have free time to find the issue, maybe on November I can work on this.

vmeleshko, 2022-10-25:
I have laptop IdeaPad 5 Pro 16ARH7 - Type 82SN and I am trying to get TP-Link TL-WN823N V3 wireless adapter working.

Laptop params:
```
Processor
1x AMD Ryzen™ 7 6800HS Processor(Ryzen™ 7 6800HS)
Memory
1x 16GBLPDDR5-6400
Hard Drive
1x 1TBPCIeGen4x4
Wireless Network
1x Bluetooth® 5.1 or above; Wi-Fi 6 2x2 AX
Ports
1x 1 x USB Type C (Full Function); 2 USB 3.2 Gen 1 (Type A); HDMI 2.0b; Audio jack; SD Card reader; 1*Type-C(PD+USB2.0)
Graphics
1x AMD Radeon™ 680M
Monitor
16" WQXGA
```

I went through the README, but unfortunately, I got the error. 

'make' all KVER=5.19.16-051916-generic...(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8192eu: 1.0 not found

```
(base) vit@idea:~$ git clone git@github.com:Mange/rtl8192eu-linux-driver.git
Cloning into 'rtl8192eu-linux-driver'...
remote: Enumerating objects: 3519, done.
remote: Counting objects: 100% (307/307), done.
remote: Compressing objects: 100% (121/121), done.
remote: Total 3519 (delta 190), reused 193 (delta 182), pack-reused 3212
Receiving objects: 100% (3519/3519), 6.86 MiB | 5.71 MiB/s, done.
Resolving deltas: 100% (2384/2384), done.
(base) vit@idea:~$ cd rtl8192eu-linux-driver/
(base) vit@idea:~/rtl8192eu-linux-driver$ sudo dkms add .
[sudo] password for vit: 
Error! DKMS tree already contains: rtl8192eu-1.0
You cannot add the same module/version combo more than once.
(base) vit@idea:~/rtl8192eu-linux-driver$ sudo dkms remove -m rtl8192eu/1.0 --all
Deleting module rtl8192eu-1.0 completely from the DKMS tree.
(base) vit@idea:~/rtl8192eu-linux-driver$ sudo dkms add .
Creating symlink /var/lib/dkms/rtl8192eu/1.0/source -> /usr/src/rtl8192eu-1.0
(base) vit@idea:~/rtl8192eu-linux-driver$ sudo dkms install rtl8192eu/1.0

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
'make' all KVER=5.19.16-051916-generic...(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8192eu: 1.0 not found
Error! Bad return status for module build on kernel: 5.19.16-051916-generic (x86_64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
(base) vit@idea:~/rtl8192eu-linux-driver$ sudo rmmod 8192eu
sudo rmmod rtl8xxxu
sudo dkms remove -m rtl8192eu -v 1.0
rmmod: ERROR: Module 8192eu is not currently loaded
rmmod: ERROR: Module rtl8xxxu is not currently loaded
Error! There is no instance of rtl8192eu 1.0 for kernel 5.19.16-051916-generic (x86_64) located in the DKMS tree.
(base) vit@idea:~/rtl8192eu-linux-driver$ sudo dkms status
rtl8192eu/1.0: added
(base) vit@idea:~/rtl8192eu-linux-driver$ sudo dkms remove -m rtl8192eu/1.0 --all
Deleting module rtl8192eu-1.0 completely from the DKMS tree.
(base) vit@idea:~/rtl8192eu-linux-driver$ sudo dkms status
(base) vit@idea:~/rtl8192eu-linux-driver$ echo "blacklist rtl8xxxu" >> ./blacklist-rtl8xxxu.conf
sudo mv ./blacklist-rtl8xxxu.conf /etc/modprobe.d/
(base) vit@idea:~/rtl8192eu-linux-driver$ sudo cp -ar . /usr/src/rtl8192eu-1.0
(base) vit@idea:~/rtl8192eu-linux-driver$ sudo dkms add -m rtl8192eu -v 1.0
Creating symlink /var/lib/dkms/rtl8192eu/1.0/source -> /usr/src/rtl8192eu-1.0
(base) vit@idea:~/rtl8192eu-linux-driver$ sudo dkms install -m rtl8192eu -v 1.0

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
'make' all KVER=5.19.16-051916-generic...(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8192eu: 1.0 not found
Error! Bad return status for module build on kernel: 5.19.16-051916-generic (x86_64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
(base) vit@idea:~/rtl8192eu-linux-driver$ 

```

**make.log**

```                                                                                                                
DKMS make.log for rtl8192eu-1.0 for kernel 5.19.16-051916-generic (x86_64)
Tue Oct 25 11:46:24 AM CEST 2022
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.19.16-051916-generic/build M=/var/lib/dkms/rtl8192eu/1.0/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.19.16-051916-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.2.0-3ubuntu1) 12.2.0
  You are using:           gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_cmd.o
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
make[2]: *** [scripts/Makefile.build:249: /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1852: /var/lib/dkms/rtl8192eu/1.0/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.19.16-051916-generic'
make: *** [Makefile:2287: modules] Error 2
```
@CGarces can you please advice here?