Mange/rtl8192eu-linux-driver #294: "Kernel NULL pointer dereference" when using this driver

dimsuz, 2022-11-07:
Driver compiled from git (tried both `master` and `realtek-4.4` branches.

OS: Arch Linux
Kernel:
> Linux 6.0.7-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 03 Nov 2022 18:01:58 +0000 x86_64 GNU/Linux

Steps to reproduce:

1. Insert TP-LINK TL-WN821N usb adapter
2. The following log is printed and UI forever "hangs" in the "Connecting..." state

```
NetworkManager[558]: <info>  [1667775663.8921] device (wlp3s0f0u3): Activation: (wifi) connection 'RT-GPON-5906' has security, and secrets exist>
NetworkManager[558]: <info>  [1667775663.8921] Config: added 'ssid' value 'RT-GPON-5906'
NetworkManager[558]: <info>  [1667775663.8921] Config: added 'scan_ssid' value '1'
NetworkManager[558]: <info>  [1667775663.8919] device (wlp3s0f0u3): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
NetworkManager[558]: <info>  [1667775663.8921] device (wlp3s0f0u3): Activation: (wifi) connection 'RT-GPON-5906' has security, and secrets exist>
NetworkManager[558]: <info>  [1667775663.8921] Config: added 'ssid' value 'RT-GPON-5906'
NetworkManager[558]: <info>  [1667775663.8921] Config: added 'scan_ssid' value '1'
NetworkManager[558]: <info>  [1667775663.8922] Config: added 'bgscan' value 'simple:30:-70:86400'
NetworkManager[558]: <info>  [1667775663.8922] Config: added 'key_mgmt' value 'WPA-PSK WPA-PSK-SHA256 FT-PSK'
NetworkManager[558]: <info>  [1667775663.8922] Config: added 'auth_alg' value 'OPEN'
NetworkManager[558]: <info>  [1667775663.8922] Config: added 'psk' value '<hidden>'
wpa_supplicant[693]: wlp3s0f0u3: WPS-CANCEL
wpa_supplicant[693]: wlp3s0f0u3: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
wpa_supplicant[693]: wlp3s0f0u3: Trying to associate with 08:f6:06:8e:59:06 (SSID='RT-GPON-5906' freq=2447 MHz)
NetworkManager[558]: <info>  [1667775663.9597] device (wlp3s0f0u3): supplicant interface state: disconnected -> associating
NetworkManager[558]: <info>  [1667775663.9597] device (p2p-dev-wlp3s0f0u3): supplicant management interface state: disconnected -> associating
wpa_supplicant[693]: wlp3s0f0u3: Trying to associate with 08:f6:06:8e:59:06 (SSID='RT-GPON-5906' freq=2447 MHz)
NetworkManager[558]: <info>  [1667775663.9597] device (wlp3s0f0u3): supplicant interface state: disconnected -> associating
NetworkManager[558]: <info>  [1667775663.9597] device (p2p-dev-wlp3s0f0u3): supplicant management interface state: disconnected -> associating
kernel: BUG: kernel NULL pointer dereference, address: 0000000000000000
kernel: #PF: supervisor read access in kernel mode
kernel: #PF: error_code(0x0000) - not-present page
kernel: PGD 1aeb5c067 P4D 1aeb5c067 PUD 0 
kernel: Oops: 0000 [#1] PREEMPT SMP NOPTI
kernel: CPU: 7 PID: 5794 Comm: kworker/u64:47 Tainted: G        W  OE      6.0.7-arch1-1 #1 54734d35253fb4c526adcfdfa2e7225be9ec4a9a
kernel: Hardware name: Micro-Star International Co., Ltd. MS-7B89/B450M MORTAR MAX (MS-7B89), BIOS 2.80 06/10/2020
kernel: Workqueue: cfg80211 cfg80211_event_work [cfg80211]
kernel: RIP: 0010:__cfg80211_connect_result+0x1f9/0x650 [cfg80211]
kernel: Code: 20 40 84 ff 74 1b 83 f9 0e 89 c8 40 0f 95 c7 83 c1 01 48 0f a3 c2 72 bf 0f b7 53 68 66 85 d2 75 e0 41 80 8c 24 cc 00 00 00 01 <41>>
kernel: RSP: 0018:ffffb0dec2cf7db0 EFLAGS: 00010202
kernel: RAX: ffff94fa6fbf4f00 RBX: ffff94fa1e619018 RCX: 0000000000000001
kernel: RDX: 0000000000000000 RSI: ffff94fa19130428 RDI: 0000000000000001
kernel: RBP: ffffb0dec2cf7e10 R08: 0000000000000036 R09: 0000000000000000
kernel: R10: 0000000000000002 R11: 00000000e5b906e6 R12: ffff94fa19130000
kernel: R13: 0000000000000001 R14: 0000000000000000 R15: ffff94fa1e619018
kernel: FS:  0000000000000000(0000) GS:ffff95007ebc0000(0000) knlGS:0000000000000000
kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
kernel: CR2: 0000000000000000 CR3: 0000000134992000 CR4: 0000000000350ee0
kernel: Call Trace:
kernel:  <TASK>
kernel:  ? cfg80211_process_wdev_events+0x15c/0x1b0 [cfg80211 3bc9c89fdd1a86213882b8cd88cb8dca2477ddbe]
kernel:  cfg80211_process_wdev_events+0x15c/0x1b0 [cfg80211 3bc9c89fdd1a86213882b8cd88cb8dca2477ddbe]
kernel:  cfg80211_process_rdev_events+0x2a/0x40 [cfg80211 3bc9c89fdd1a86213882b8cd88cb8dca2477ddbe]
kernel:  cfg80211_event_work+0x29/0x40 [cfg80211 3bc9c89fdd1a86213882b8cd88cb8dca2477ddbe]
kernel:  process_one_work+0x1c7/0x380
kernel:  worker_thread+0x51/0x390
kernel:  ? rescuer_thread+0x3b0/0x3b0
kernel:  kthread+0xde/0x110
kernel:  ? kthread_complete_and_exit+0x20/0x20
kernel:  ret_from_fork+0x22/0x30
kernel:  </TASK>
kernel: Modules linked in: 8192eu(OE) tls mac80211 libarc4 cfg80211 rfkill nct6775 nct6775_core hwmon_vid intel_rapl_msr intel_rapl_common snd_h>
kernel: CR2: 0000000000000000
kernel: ---[ end trace 0000000000000000 ]---
kernel: RIP: 0010:__cfg80211_connect_result+0x1f9/0x650 [cfg80211]
kernel: Code: 20 40 84 ff 74 1b 83 f9 0e 89 c8 40 0f 95 c7 83 c1 01 48 0f a3 c2 72 bf 0f b7 53 68 66 85 d2 75 e0 41 80 8c 24 cc 00 00 00 01 <41>>
kernel: RSP: 0018:ffffb0dec2cf7db0 EFLAGS: 00010202
kernel: RAX: ffff94fa6fbf4f00 RBX: ffff94fa1e619018 RCX: 0000000000000001
kernel: RDX: 0000000000000000 RSI: ffff94fa19130428 RDI: 0000000000000001
kernel: RBP: ffffb0dec2cf7e10 R08: 0000000000000036 R09: 0000000000000000
kernel: R10: 0000000000000002 R11: 00000000e5b906e6 R12: ffff94fa19130000
kernel: R13: 0000000000000001 R14: 0000000000000000 R15: ffff94fa1e619018
kernel: FS:  0000000000000000(0000) GS:ffff95007ebc0000(0000) knlGS:0000000000000000
kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
kernel: CR2: 0000000000000000 CR3: 0000000134992000 CR4: 0000000000350ee0
```

pterjan, 2022-11-16:
Can you verify if https://github.com/Mange/rtl8192eu-linux-driver/pull/296 fixes this one too?

dimsuz, 2022-12-28:
Sorry for the long answer @pterjan , somehow I missed your reply.

Looks like "NULL pointer dereference" is gone, but the other error is printed now. It seems not fatal, but in case it's somehow related (I know nothing about kernel development), I'll post it here, you decide if this issue should be closed now:

```
kernel: ------------[ cut here ]------------
kernel: memcpy: detected field-spanning write (size 16) of single field "ht_capie.mcs.rx_mask" at /home/dima/.cache/yay/rtl8192eu-git/src/rtl8192eu-linux-driver/core/rtw_mlme.c:4744 (size 10)
kernel: WARNING: CPU: 2 PID: 698 at /home/dima/.cache/yay/rtl8192eu-git/src/rtl8192eu-linux-driver/core/rtw_mlme.c:4744 rtw_restructure_ht_ie+0x4ca/0x4f0 [8192eu]
kernel: Modules linked in: 8192eu(OE) mac80211 libarc4 cfg80211 snd_seq_dummy snd_seq snd_seq_device ntfs3 rfkill nct6775 nct6775_core intel_rapl_msr hwmon_vid intel_rapl_common edac_mce_amd snd_hda_codec_realtek kvm_amd snd_hda_codec_generic mousedev ledtrig_audio snd_hda_codec_hdmi joydev ppdev snd_hda_intel kvm snd_intel_dspcfg snd_intel_sdw_acpi nls_iso8859_1 irqbypass crct10dif_pclmul vfat snd_hda_codec crc32_pclmul fat snd_hda_core polyval_clmulni polyval_generic gf128mul snd_hwdep ghash_clmulni_intel snd_pcm sha512_ssse3 aesni_intel r8169 snd_timer uas crypto_simd snd usb_storage cryptd usbhid realtek rapl mdio_devres soundcore wmi_bmof sp5100_tco ccp libphy pcspkr i2c_piix4 k10temp parport_pc gpio_amdpt parport gpio_generic mac_hid acpi_cpufreq crypto_user fuse bpf_preload ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 crc32c_intel xhci_pci xhci_pci_renesas amdgpu drm_ttm_helper ttm video wmi gpu_sched drm_buddy drm_display_helper cec [last unloaded: rtl8xxxu]
kernel: CPU: 2 PID: 698 Comm: wpa_supplicant Tainted: G           OE      6.1.1-arch1-1 #1 9bd09188b430be630e611f984454e4f3c489be77
kernel: Hardware name: Micro-Star International Co., Ltd. MS-7B89/B450M MORTAR MAX (MS-7B89), BIOS 2.80 06/10/2020
kernel: RIP: 0010:rtw_restructure_ht_ie+0x4ca/0x4f0 [8192eu]
kernel: Code: ff e9 00 fe ff ff b9 0a 00 00 00 48 c7 c2 40 02 e3 c1 be 10 00 00 00 48 c7 c7 90 00 e3 c1 c6 05 22 e7 13 00 01 e8 f1 9b 25 e7 <0f> 0b e9 91 fc ff ff 41 80 bf bd 07 00 00 00 0f 84 01 fc ff ff eb
kernel: RSP: 0018:ffffb29c02843278 EFLAGS: 00010282
kernel: RAX: 0000000000000000 RBX: 000000000000002b RCX: 0000000000000027
kernel: RDX: ffff90fafeaa1668 RSI: 0000000000000001 RDI: ffff90fafeaa1660
kernel: RBP: ffff90f7371428a0 R08: 0000000000000000 R09: ffffb29c02843100
kernel: R10: 0000000000000003 R11: ffff90fb1f328c28 R12: ffffb29c017bdd18
kernel: R13: ffff90f7371428a4 R14: ffffb29c003eb7da R15: ffffb29c003eb000
kernel: FS:  00007f4298452c00(0000) GS:ffff90fafea80000(0000) knlGS:0000000000000000
kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
kernel: CR2: 00003e82ddd5c000 CR3: 000000012b732000 CR4: 0000000000350ee0
kernel: Call Trace:
kernel:  <TASK>
kernel:  rtw_joinbss_cmd+0x3de/0x420 [8192eu 459b89738815070d3960e19033824751f26c581c]
kernel:  rtw_select_and_join_from_scanned_queue+0x62/0x1f0 [8192eu 459b89738815070d3960e19033824751f26c581c]
kernel:  rtw_do_join+0x10a/0x260 [8192eu 459b89738815070d3960e19033824751f26c581c]
kernel:  rtw_set_802_11_connect+0x11f/0x1c0 [8192eu 459b89738815070d3960e19033824751f26c581c]
kernel:  cfg80211_rtw_set_default_mgmt_key+0x2c44/0x4230 [8192eu 459b89738815070d3960e19033824751f26c581c]
kernel:  cfg80211_connect+0x192/0x780 [cfg80211 5e9ae944bff665674dd710fdea59d3a8ecca8922]
kernel:  ? nl80211_prepare_wdev_dump+0xd7/0x210 [cfg80211 5e9ae944bff665674dd710fdea59d3a8ecca8922]
kernel:  nl80211_connect+0x558/0x720 [cfg80211 5e9ae944bff665674dd710fdea59d3a8ecca8922]
kernel:  genl_family_rcv_msg_doit+0x100/0x160
kernel:  genl_rcv_msg+0x126/0x250
kernel:  ? nl80211_parse_connkeys+0x2e0/0x2e0 [cfg80211 5e9ae944bff665674dd710fdea59d3a8ecca8922]
kernel:  ? genl_start+0x170/0x170
kernel:  netlink_rcv_skb+0x55/0x100
kernel:  genl_rcv+0x28/0x40
kernel:  netlink_unicast+0x246/0x390
kernel:  netlink_sendmsg+0x254/0x4d0
kernel:  sock_sendmsg+0x63/0x70
kernel:  ____sys_sendmsg+0x277/0x2f0
kernel:  ? copy_msghdr_from_user+0x7d/0xc0
kernel:  ___sys_sendmsg+0x9a/0xe0
kernel:  __sys_sendmsg+0x7a/0xd0
kernel:  do_syscall_64+0x5f/0x90
kernel:  ? syscall_exit_to_user_mode+0x1b/0x40
kernel:  ? do_syscall_64+0x6b/0x90
kernel:  ? do_syscall_64+0x6b/0x90
kernel:  ? do_syscall_64+0x6b/0x90
kernel:  entry_SYSCALL_64_after_hwframe+0x63/0xcd
kernel: RIP: 0033:0x7f4297f22da4
kernel: Code: 15 e9 df 0c 00 f7 d8 64 89 02 48 c7 c0 ff ff ff ff eb b7 0f 1f 00 f3 0f 1e fa 80 3d 7d 67 0d 00 00 74 13 b8 2e 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 54 c3 0f 1f 00 48 83 ec 28 89 54 24 1c 48 89
kernel: RSP: 002b:00007ffc5c8880b8 EFLAGS: 00000202 ORIG_RAX: 000000000000002e
kernel: RAX: ffffffffffffffda RBX: 00005652e1be6c70 RCX: 00007f4297f22da4
kernel: RDX: 0000000000000000 RSI: 00007ffc5c8880f0 RDI: 0000000000000006
kernel: RBP: 00005652e1c546b0 R08: 0000000000000004 R09: 0000000000000000
kernel: R10: 00007ffc5c8881d0 R11: 0000000000000202 R12: 00005652e1be74c0
kernel: R13: 00007ffc5c8880f0 R14: 0000000000000000 R15: 00007ffc5c8881d0
kernel:  </TASK>
kernel: ---[ end trace 0000000000000000 ]---
```

dimsuz, 2022-12-28:
oh, wait. Right after I posted the above I re-plugged the adapter and got the null ptr deref:

```
NetworkManager[549]: <info>  [1672187417.3916] device (wlp3s0f0u3): supplicant interface state: interface_disabled -> inactive
NetworkManager[549]: <info>  [1672187417.3916] device (p2p-dev-wlp3s0f0u3): supplicant management interface state: interface_disabled -> inactive
wpa_supplicant[698]: wlp3s0f0u3: Trying to associate with 08:f6:06:8e:59:06 (SSID='RT-GPON-5906' freq=2442 MHz)
NetworkManager[549]: <info>  [1672187417.4042] device (wlp3s0f0u3): supplicant interface state: inactive -> associating
NetworkManager[549]: <info>  [1672187417.4042] device (p2p-dev-wlp3s0f0u3): supplicant management interface state: inactive -> associating
kernel: BUG: kernel NULL pointer dereference, address: 0000000000000000
kernel: #PF: supervisor read access in kernel mode
kernel: #PF: error_code(0x0000) - not-present page
kernel: PGD 1c38d9067 P4D 1c38d9067 PUD 0 
kernel: Oops: 0000 [#1] PREEMPT SMP NOPTI
kernel: CPU: 1 PID: 10366 Comm: kworker/u64:0 Tainted: G        W  OE      6.1.1-arch1-1 #1 9bd09188b430be630e611f984454e4f3c489be77
kernel: Hardware name: Micro-Star International Co., Ltd. MS-7B89/B450M MORTAR MAX (MS-7B89), BIOS 2.80 06/10/2020
kernel: Workqueue: cfg80211 cfg80211_event_work [cfg80211]
kernel: RIP: 0010:__cfg80211_connect_result+0x208/0x670 [cfg80211]
kernel: Code: 20 40 84 ff 74 1b 83 f9 0e 89 c8 40 0f 95 c7 83 c1 01 48 0f a3 c2 72 bf 0f b7 53 68 66 85 d2 75 e0 41 80 8c 24 cc 00 00 00 01 <41> 8b 06 41 89 84 24 b8 03 00 00 41 0f b7 46 04 66 41 89 84 24 bc
kernel: RSP: 0018:ffffb29c09233db0 EFLAGS: 00010202
kernel: RAX: ffff90f67055d800 RBX: ffff90f8e7714818 RCX: 0000000000000001
kernel: RDX: 0000000000000000 RSI: ffff90f4164c7428 RDI: 0000000000000001
kernel: RBP: ffffb29c09233e10 R08: 0000000000000036 R09: 0000000000000000
kernel: R10: 0000000000000002 R11: 00000000a5fed6a9 R12: ffff90f4164c7000
kernel: R13: 0000000000000001 R14: 0000000000000000 R15: ffff90f8e7714818
kernel: FS:  0000000000000000(0000) GS:ffff90fafea40000(0000) knlGS:0000000000000000
kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
kernel: CR2: 0000000000000000 CR3: 0000000209db4000 CR4: 0000000000350ee0
kernel: Call Trace:
kernel:  <TASK>
kernel:  ? psi_task_switch+0xd6/0x230
kernel:  ? cfg80211_process_wdev_events+0x15c/0x1b0 [cfg80211 5e9ae944bff665674dd710fdea59d3a8ecca8922]
kernel:  cfg80211_process_wdev_events+0x15c/0x1b0 [cfg80211 5e9ae944bff665674dd710fdea59d3a8ecca8922]
kernel:  cfg80211_process_rdev_events+0x2a/0x40 [cfg80211 5e9ae944bff665674dd710fdea59d3a8ecca8922]
kernel:  cfg80211_event_work+0x29/0x40 [cfg80211 5e9ae944bff665674dd710fdea59d3a8ecca8922]
kernel:  process_one_work+0x1c7/0x380
kernel:  worker_thread+0x51/0x390
kernel:  ? rescuer_thread+0x3b0/0x3b0
kernel:  kthread+0xde/0x110
kernel:  ? kthread_complete_and_exit+0x20/0x20
kernel:  ret_from_fork+0x22/0x30
kernel:  </TASK>
kernel: Modules linked in: 8192eu(OE) mac80211 libarc4 cfg80211 snd_seq_dummy snd_seq snd_seq_device ntfs3 rfkill nct6775 nct6775_core intel_rapl_msr hwmon_vid intel_rapl_common edac_mce_amd snd_hda_codec_realtek kvm_amd snd_hda_codec_generic mousedev ledtrig_audio snd_hda_codec_hdmi joydev ppdev snd_hda_intel kvm snd_intel_dspcfg snd_intel_sdw_acpi nls_iso8859_1 irqbypass crct10dif_pclmul vfat snd_hda_codec crc32_pclmul fat snd_hda_core polyval_clmulni polyval_generic gf128mul snd_hwdep ghash_clmulni_intel snd_pcm sha512_ssse3 aesni_intel r8169 snd_timer uas crypto_simd snd usb_storage cryptd usbhid realtek rapl mdio_devres soundcore wmi_bmof sp5100_tco ccp libphy pcspkr i2c_piix4 k10temp parport_pc gpio_amdpt parport gpio_generic mac_hid acpi_cpufreq crypto_user fuse bpf_preload ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 crc32c_intel xhci_pci xhci_pci_renesas amdgpu drm_ttm_helper ttm video wmi gpu_sched drm_buddy drm_display_helper cec [last unloaded: rtl8xxxu]
kernel: CR2: 0000000000000000
kernel: ---[ end trace 0000000000000000 ]---
kernel: RIP: 0010:__cfg80211_connect_result+0x208/0x670 [cfg80211]
kernel: Code: 20 40 84 ff 74 1b 83 f9 0e 89 c8 40 0f 95 c7 83 c1 01 48 0f a3 c2 72 bf 0f b7 53 68 66 85 d2 75 e0 41 80 8c 24 cc 00 00 00 01 <41> 8b 06 41 89 84 24 b8 03 00 00 41 0f b7 46 04 66 41 89 84 24 bc
kernel: RSP: 0018:ffffb29c09233db0 EFLAGS: 00010202
kernel: RAX: ffff90f67055d800 RBX: ffff90f8e7714818 RCX: 0000000000000001
kernel: RDX: 0000000000000000 RSI: ffff90f4164c7428 RDI: 0000000000000001
kernel: RBP: ffffb29c09233e10 R08: 0000000000000036 R09: 0000000000000000
kernel: R10: 0000000000000002 R11: 00000000a5fed6a9 R12: ffff90f4164c7000
kernel: R13: 0000000000000001 R14: 0000000000000000 R15: ffff90f8e7714818
kernel: FS:  0000000000000000(0000) GS:ffff90fafea40000(0000) knlGS:0000000000000000
kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
kernel: CR2: 0000000000000000 CR3: 0000000209db4000 CR4: 0000000000350ee0
```

dimsuz, 2022-12-28:
This was on commit https://github.com/Mange/rtl8192eu-linux-driver/commit/aeed529c77282cc0d8a2b6eb29186711e25555ce (includes the PR you've linked), and this kernel version:

```
Linux 6.1.1-arch1-1 #1 SMP PREEMPT_DYNAMIC Wed, 21 Dec 2022 22:27:55 +0000 x86_64 GNU/Linux
```

nicbn, 2023-03-06:
Also experiencing this. Fedora 37, kernel 6.1.14-200

kkkaamis, 2023-10-13:
Having this issue as well, debian 12

Diadlo, 2025-05-14:
Have the same issue. Arch Linux, kernel 6.14.5-arch1-1