Mange/rtl8192eu-linux-driver #306: error on ubuntu 22.10 kernel 6.1.4 and 6.1.5

rachmadaniHaryono, 2023-01-13:
error on ubuntu 22.10 kernel 6.1.4 and 6.1.5

with https://github.com/Mange/rtl8192eu-linux-driver/commit/aeed529c77282cc0d8a2b6eb29186711e25555ce
possible highest running kernel from mainline is 6.0.19

<details><summary>make.log</summary>
<p>

```c
DKMS make.log for rtl8192eu-1.0 for kernel 6.1.5-060105-generic (x86_64)
Jum 13 Jan 2023 06:40:29  WITA
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.1.5-060105-generic/build M=/var/lib/dkms/rtl8192eu/1.0/build  modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.5-060105-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.2.0-10ubuntu1) 12.2.0
  You are using:           gcc-12 (Ubuntu 12.2.0-3ubuntu1) 12.2.0
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme.o
In file included from ./include/linux/string.h:253,
                 from ./include/linux/bitmap.h:11,
                 from ./include/linux/cpumask.h:12,
                 from ./arch/x86/include/asm/cpumask.h:5,
                 from ./arch/x86/include/asm/msr.h:11,
                 from ./arch/x86/include/asm/processor.h:22,
                 from ./arch/x86/include/asm/timex.h:5,
                 from ./include/linux/timex.h:67,
                 from ./include/linux/time32.h:13,
                 from ./include/linux/time.h:60,
                 from ./include/linux/stat.h:19,
                 from ./include/linux/module.h:13,
                 from /var/lib/dkms/rtl8192eu/1.0/build/include/basic_types.h:37,
                 from /var/lib/dkms/rtl8192eu/1.0/build/include/drv_types.h:26,
                 from /var/lib/dkms/rtl8192eu/1.0/build/include/../hal/phydm/phydm_types.h:215,
                 from /var/lib/dkms/rtl8192eu/1.0/build/include/../hal/phydm/phydm_precomp.h:29,
                 from /var/lib/dkms/rtl8192eu/1.0/build/include/hal_data.h:20,
                 from /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme.c:17:
In function ‘fortify_memcpy_chk’,
    inlined from ‘rtw_restructure_ht_ie’ at /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme.c:4744:2:
./include/linux/fortify-string.h:404:25: warning: call to ‘__write_overflow_field’ declared with attribute warning: detected write beyond size of field (1st parameter); maybe use struct_group()? [-Wattribute-warning]
  404 |                         __write_overflow_field(p_size_field, size);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.o
/var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.c: In function ‘rtw_random32’:
/var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.c:1621:16: error: too few arguments to function ‘prandom_u32_max’
 1621 |         return prandom_u32_max();
      |                ^~~~~~~~~~~~~~~
In file included from ./include/linux/random.h:172,
                 from ./include/linux/nodemask.h:97,
                 from ./include/linux/mmzone.h:17,
                 from ./include/linux/gfp.h:7,
                 from ./include/linux/umh.h:4,
                 from ./include/linux/kmod.h:9,
                 from ./include/linux/module.h:17,
                 from /var/lib/dkms/rtl8192eu/1.0/build/include/basic_types.h:37,
                 from /var/lib/dkms/rtl8192eu/1.0/build/include/drv_types.h:26,
                 from /var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.c:19:
./include/linux/prandom.h:27:19: note: declared here
   27 | static inline u32 prandom_u32_max(u32 ep_ro)
      |                   ^~~~~~~~~~~~~~~
/var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.c:1633:1: error: control reaches end of non-void function [-Werror=return-type]
 1633 | }
      | ^
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.o] Error 1
make[1]: *** [Makefile:1998: /var/lib/dkms/rtl8192eu/1.0/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.5-060105-generic'
make: *** [Makefile:2287: modules] Error 2
```

</p>
</details>

CGarces, 2023-01-13:
Hi.

Compilations looks fine on CI
https://github.com/Mange/rtl8192eu-linux-driver/actions/runs/3777144815/jobs/6684467382

The issue with prandom_u32_max was fixed on #290

I don't know why prandom_u32_max is called

rachmadaniHaryono, 2023-01-13:
gcc version

```console
❯ gcc --version
gcc (Ubuntu 12.2.0-3ubuntu1) 12.2.0
```

https://github.com/Mange/rtl8192eu-linux-driver/actions/runs/3777144815/jobs/6684466726 this one is for 6.1.5 with gcc 12

i don't know how to continue with this. without it i can't connect to internet

i will try to reinstall module and report back if possible

if not i will wait for next kernel version

rachmadaniHaryono, 2023-01-22:
i think i may have fix it

first check if `/var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.c` is the same as the file on this repo

if not try remove and reinstall the module (not sure if build is required, but i did it anyway

```console
sudo dkms remove rtl8192eu/1.0
sudo dkms build rtl8192eu/1.0
sudo dkms install rtl8192eu/1.0
```

i haven't check the two kernels but 6.1.7 is working

closed 2023-01-22 by rachmadaniHaryono