Mange/rtl8192eu-linux-driver #307: compilation issue on arch linux kernel 6.1.6-arch1-3

nalin243, 2023-01-21:
[make.log](https://github.com/Mange/rtl8192eu-linux-driver/files/10471538/make.log)

criticic, 2023-01-22:
Facing same issue!

criticic, 2023-01-23:
fixed. in my case all i had to do was update to 6.1.6. Related #306

glibg10b, 2023-03-05:
I'm running kernel version 6.1.12 and linux-headers version 6.2, but I'm getting the same error on Gentoo when building the package from the GURU repo:

```text
  CC [M]  /var/tmp/portage/net-wireless/rtl8192eu-0_pre20220825/work/rtl8192eu-linux-driver-e39c4e7a66b05fab6eceb2bb251d399c003fb544/os_dep/linux/xmit_linux.o
/var/tmp/portage/net-wireless/rtl8192eu-0_pre20220825/work/rtl8192eu-linux-driver-e39c4e7a66b05fab6eceb2bb251d399c003fb544/os_dep/osdep_service.c: In function ‘rtw_random32’:
/var/tmp/portage/net-wireless/rtl8192eu-0_pre20220825/work/rtl8192eu-linux-driver-e39c4e7a66b05fab6eceb2bb251d399c003fb544/os_dep/osdep_service.c:1622:16: error: implicit declaration of function ‘prandom_u32’; did you mean ‘prandom_u32_max’? [-Werror=implicit-function-declaration]
 1622 |         return prandom_u32();
      |                ^~~~~~~~~~~
      |                prandom_u32_max
  CC [M]  /var/tmp/portage/net-wireless/rtl8192eu-0_pre20220825/work/rtl8192eu-linux-driver-e39c4e7a66b05fab6eceb2bb251d399c003fb544/os_dep/linux/mlme_linux.o
/var/tmp/portage/net-wireless/rtl8192eu-0_pre20220825/work/rtl8192eu-linux-driver-e39c4e7a66b05fab6eceb2bb251d399c003fb544/os_dep/linux/os_intfs.c: In function ‘rtw_os_ndev_register’:
/var/tmp/portage/net-wireless/rtl8192eu-0_pre20220825/work/rtl8192eu-linux-driver-e39c4e7a66b05fab6eceb2bb251d399c003fb544/os_dep/linux/os_intfs.c:1676:9: error: too many arguments to function ‘netif_napi_add’
 1676 |         netif_napi_add(ndev, &adapter->napi, rtw_recv_napi_poll, RTL_NAPI_WEIGHT);
      |         ^~~~~~~~~~~~~~
In file included from /var/tmp/portage/net-wireless/rtl8192eu-0_pre20220825/work/rtl8192eu-linux-driver-e39c4e7a66b05fab6eceb2bb251d399c003fb544/include/osdep_service_linux.h:30,
                 from /var/tmp/portage/net-wireless/rtl8192eu-0_pre20220825/work/rtl8192eu-linux-driver-e39c4e7a66b05fab6eceb2bb251d399c003fb544/include/osdep_service.h:45,
                 from /var/tmp/portage/net-wireless/rtl8192eu-0_pre20220825/work/rtl8192eu-linux-driver-e39c4e7a66b05fab6eceb2bb251d399c003fb544/include/drv_types.h:27,
                 from /var/tmp/portage/net-wireless/rtl8192eu-0_pre20220825/work/rtl8192eu-linux-driver-e39c4e7a66b05fab6eceb2bb251d399c003fb544/os_dep/linux/os_intfs.c:17:
./include/linux/netdevice.h:2569:1: note: declared here
 2569 | netif_napi_add(struct net_device *dev, struct napi_struct *napi,
      | ^~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:250: /var/tmp/portage/net-wireless/rtl8192eu-0_pre20220825/work/rtl8192eu-linux-driver-e39c4e7a66b05fab6eceb2bb251d399c003fb544/os_dep/linux/os_intfs.o] Error 1
```

CGarces, 2023-03-05:
The issue with prandom_u32_max was fixed on https://github.com/Mange/rtl8192eu-linux-driver/pull/290
Duplicate of #306

closed 2023-03-05 by CGarces