Mange/rtl8192eu-linux-driver #313: POR FAVOR ME PUEDEN AYUDAR PARA INSTALAR USB WIRELESS TP LINK TL-WN821N EN MODO MONITOR

Chucho-C, 2023-02-27:


albaluz35, 2024-05-16:
In my case enabling monitor mode and doing packet injection WORKS! as follows:

sudo apt update
sudo apt upgrade
sudo apt-get install linux-headers-`uname -r` build-essential dkms
git clone https://github.com/clnhub/rtl8192eu-linux.git
cd rtl8192eu-linux
sed -i -E 's/CONFIG_WIFI_MONITOR\s*=\s*n/CONFIG_WIFI_MONITOR = y/' Makefile

sudo rmmod 8192eu rtl8xxxu r8188eu
echo "blacklist rtl8xxxu" >> ./blacklist-rtl8xxxu.conf
sudo mv ./blacklist-rtl8xxxu.conf /etc/modprobe.d/
./install_wifi.sh

-----------------------

**IF THERE ARE PROBLEMS WITH ./install_wifi.sh, RUN ./uninstall_wifi.sh then execute make commands manually:**
\# sudo make
\# sudo make install

-----------------

sudo modprobe 8192eu
sudo lsmod |grep 8192eu                                                     

!To enable Monitor mode and test packet injection:!
=================================================

WL=wlan0
sudo iwconfig
sudo ifconfig $WL down
sudo airmon-ng check kill
sudo iwconfig $WL mode monitor
sudo ifconfig $WL up
sudo iwconfig

\# Wait for a moment and retry if there are issues or reboot and try the "
!To enable Monitor mode and test packet injection:! " steps
sudo aireplay-ng --test $WL

CGarces, 2024-05-18:
thanks @albaluz35

closed 2024-05-18 by CGarces