Mange/rtl8192eu-linux-driver #317: Can't compile driver

stefanlk, 2023-04-02:
when compiling on my Raspi I got following messages:

>dkms install rtl8192eu/1.0

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=6.1.19-v7+...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.19-v7+ (armv7l)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
> more /var/lib/dkms/rtl8192eu/1.0/build/make.log
DKMS make.log for rtl8192eu-1.0 for kernel 6.1.19-v7+ (armv7l)
So 2. Apr 12:19:43 CEST 2023
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/6.1.19-v7+/build M=/var/lib/dkms/rtl8192eu/1.0/build  modules
make[1]: Verzeichnis ▒/usr/src/linux-headers-6.1.19-v7+▒ wird betreten
Makefile:775: arch/armv7l/Makefile: Datei oder Verzeichnis nicht gefunden
make[1]: *** Keine Regel, um ▒arch/armv7l/Makefile▒ zu erstellen.  Schluss.
make[1]: Verzeichnis ▒/usr/src/linux-headers-6.1.19-v7+▒ wird verlassen
make: *** [Makefile:2287: modules] Fehler 2

>uname -r
6.1.19-v7+
> uname -a
Linux raspi-family 6.1.19-v7+ #1637 SMP Tue Mar 14 11:04:52 GMT 2023 armv7l GNU/Linux

> ls -la /usr/src
insgesamt 28
drwxr-xr-x  7 root root 4096  2. Apr 11:34 .
drwxr-xr-x 22 root root 4096  8. Dez 2021  ..
lrwxrwxrwx  1 root root   24  2. Apr 11:33 linux -> linux-headers-6.1.19-v7+
drwxr-xr-x 25 root root 4096  2. Apr 11:31 linux-headers-6.1.19+
drwxr-xr-x 25 root root 4096  2. Apr 12:14 linux-headers-6.1.19-v7+
drwxr-xr-x 25 root root 4096  2. Apr 11:31 linux-headers-6.1.19-v7l+
drwxr-xr-x  7 root root 4096  2. Apr 11:34 rtl8192eu-1.0
drwxr-xr-x  3 root root 4096  4. M▒r 2021  sense-hat

whats the problem?
why does the make says architect armv7l? Shall it be arm?

Dschogo, 2023-08-11:
If the rpi is running 64bit use, its not called rpi but works for me (kernel 6.1.21-b8+ - rpi3b+)
`CONFIG_PLATFORM_ARM_RPI = n`
`CONFIG_PLATFORM_ARM_AARCH64 = y` 

<details>
  <summary>Click me for Makefile</summary>

```make
EXTRA_CFLAGS += $(USER_EXTRA_CFLAGS)

##################### Optimization Flags ##########################
EXTRA_CFLAGS += -O2
##################### Compile-time Warnings #######################
EXTRA_CFLAGS += -Wno-unused-variable
EXTRA_CFLAGS += -Wno-unused-label
EXTRA_CFLAGS += -Wno-unused-function

GCC_VER_49 := $(shell echo `$(CC) -dumpversion | cut -f1-2 -d.` \>= 4.9 | bc )
ifeq ($(GCC_VER_49),1)
endif

EXTRA_CFLAGS += -I$(src)/include

EXTRA_LDFLAGS += --strip-debug

CONFIG_AUTOCFG_CP = n

########################## WIFI IC ############################
CONFIG_MULTIDRV = n
CONFIG_RTL8188E = n
CONFIG_RTL8812A = n
CONFIG_RTL8821A = n
CONFIG_RTL8192E = y
CONFIG_RTL8723B = n
CONFIG_RTL8814A = n
CONFIG_RTL8723C = n
CONFIG_RTL8188F = n
CONFIG_RTL8188GTV = n
CONFIG_RTL8822B = n
CONFIG_RTL8723D = n
CONFIG_RTL8821C = n
CONFIG_RTL8710B = n
CONFIG_RTL8192F = n
######################### Interface ###########################
CONFIG_USB_HCI = y
CONFIG_PCI_HCI = n
CONFIG_SDIO_HCI = n
CONFIG_GSPI_HCI = n
########################## Features ###########################
CONFIG_MP_INCLUDED = y
# Exposes power saving features to the system if the wireless adapter supports them.
# Say "y" here if unsure.
CONFIG_POWER_SAVING = y
# Automatically suspends the wireless adapter during inactivity.
# Say "n" here if unsure. It may save a small amount of power, while introducing instability.
CONFIG_USB_AUTOSUSPEND = n
CONFIG_HW_PWRP_DETECTION = n
CONFIG_BT_COEXIST = y
CONFIG_INTEL_WIDI = n
CONFIG_WAPI_SUPPORT = n
CONFIG_EFUSE_CONFIG_FILE = y
CONFIG_EXT_CLK = n
CONFIG_TRAFFIC_PROTECT = n
CONFIG_LOAD_PHY_PARA_FROM_FILE = y
CONFIG_TXPWR_BY_RATE = y
CONFIG_TXPWR_BY_RATE_EN = n
CONFIG_TXPWR_LIMIT = y
CONFIG_TXPWR_LIMIT_EN = n
CONFIG_RTW_CHPLAN = 0xFF
CONFIG_RTW_ADAPTIVITY_EN = disable
[... 104 lines cut ...]
CONFIG_PLATFORM_ARM_SPREADTRUM_8810 = n
CONFIG_PLATFORM_ARM_WMT = n
CONFIG_PLATFORM_TI_DM365 = n
CONFIG_PLATFORM_MOZART = n
CONFIG_PLATFORM_RTK119X = n
CONFIG_PLATFORM_RTK119X_AM = n
CONFIG_PLATFORM_RTK129X = n
CONFIG_PLATFORM_RTK390X = n
CONFIG_PLATFORM_NOVATEK_NT72668 = n
CONFIG_PLATFORM_HISILICON = n
CONFIG_PLATFORM_HISILICON_HI3798 = n
CONFIG_PLATFORM_NV_TK1 = n
CONFIG_PLATFORM_NV_TK1_UBUNTU = n
CONFIG_PLATFORM_RTL8197D = n
CONFIG_PLATFORM_AML_S905 = n
CONFIG_PLATFORM_ZTE_ZX296716 = n

########### CUSTOMER ################################
CONFIG_CUSTOMER_HUAWEI_GENERAL = n

CONFIG_DRVEXT_MODULE = n

export TopDIR ?= $(shell pwd)
```
</details>

CGarces, 2024-01-20:
@stefanlk use suggestion from last comment.

closed 2024-01-20 by CGarces