Mange/rtl8192eu-linux-driver #323: DKMS Build issue on Debian 11.7

fnatter, 2023-08-26:
hello community,

I cannot add the module with "sudo dkms add ." because I have an existing rtl8192eu/1.0
module that works for older D11 kernels (compatibility broke with 5.10.0-25).
So I want to keep rtl8192eu for older kernels (like 5.10.0-23), that's why
I don't want to use "sudo dkms remove rtl8192/1.0 --all".

Question 1: If I build using 
 sudo dkms build rtl8192/1.0 -k 5.10.0-25-amd64
does it build the new module (working directory) instead of the one
that was installed using "sudo dkms add ."?
Can I remove the module rtl8192/1.0 *for* *this* *kernel* so that I can
"sudo dkms add ." it?

Question 2: I get the same issue as in #309 (BUT please see Question 1).

Many Thanks and Best Regards,
Felix

a5a5aa555oo, 2023-08-29:
Before this issue is fixed, you can use rtl8xxxu instead.

https://github.com/a5a5aa555oo/rtl8xxxu

fnatter, 2023-09-02:
hello a5a5aa555oo,
many thanks, "modprobe rtl8xxxu" did the trick!
Best Regards,
Felix

closed 2023-09-02 by fnatter