Mange/rtl8192eu-linux-driver #329: Kernel error

kkkaamis, 2023-11-18:
Hi,
I've been trying to get the adapter TL-WN821N V6 to work on Debian for some time now, but I don't have much experience.
I'm getting this error, no idea what it is but i'll leave it here if someone knows.
The driver works half the time, but the other half of the time it freezes and I have to reboot.
Kernel is 6.4.0-0.deb12.2-amd64

```
------------[ cut here ]------------
nov 19 00:28:46 ratinkreesls kernel: memcpy: detected field-spanning write (size 16) of single field "ht_capie.mcs.rx_mask" at /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme.c:4744 (size 10)
nov 19 00:28:46 ratinkreesls kernel: WARNING: CPU: 7 PID: 844 at /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme.c:4744 rtw_restructure_ht_ie+0x4e2/0x510 [8192eu]
nov 19 00:28:46 ratinkreesls kernel: Modules linked in: vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) qrtr nvidia_drm(POE) nvidia_modeset(POE) binfmt_misc squashfs nvidia(POE) intel_rapl_msr intel_ra>
nov 19 00:28:46 ratinkreesls kernel:  crct10dif_generic scsi_mod dca i2c_piix4 i2c_algo_bit crct10dif_pclmul usb_common scsi_common crct10dif_common wmi gpio_amdpt gpio_generic button
nov 19 00:28:46 ratinkreesls kernel: CPU: 7 PID: 844 Comm: wpa_supplicant Tainted: P           OE      6.4.0-0.deb12.2-amd64 #1  Debian 6.4.4-3~bpo12+1
nov 19 00:28:46 ratinkreesls kernel: Hardware name: System manufacturer System Product Name/PRIME X370-PRO, BIOS 6203 07/27/2023
nov 19 00:28:46 ratinkreesls kernel: RIP: 0010:rtw_restructure_ht_ie+0x4e2/0x510 [8192eu]
nov 19 00:28:46 ratinkreesls kernel: Code: 00 e9 1f fd ff ff b9 0a 00 00 00 48 c7 c2 18 ec e9 c0 be 10 00 00 00 48 c7 c7 c0 ea e9 c0 c6 05 6a c3 0e 00 01 e8 2e f8 57 ec <0f> 0b e9 79 fc ff ff 41 >
nov 19 00:28:46 ratinkreesls kernel: RSP: 0018:ffffa4714146b298 EFLAGS: 00010296
nov 19 00:28:46 ratinkreesls kernel: RAX: 0000000000000000 RBX: 000000000000002b RCX: 0000000000000000
nov 19 00:28:46 ratinkreesls kernel: RDX: 0000000000000404 RSI: ffffffffae440eea RDI: 00000000ffffffff
nov 19 00:28:46 ratinkreesls kernel: RBP: ffff971c128aa0a0 R08: 0000000000000000 R09: ffffa4714146b128
nov 19 00:28:46 ratinkreesls kernel: R10: 0000000000000003 R11: ffff971d2f3048a8 R12: ffffa471413b9c48
nov 19 00:28:46 ratinkreesls kernel: R13: ffff971c128aa0a4 R14: ffffa47140df97da R15: ffffa47140df9000
nov 19 00:28:46 ratinkreesls kernel: FS:  00007f248de98c80(0000) GS:ffff971d26bc0000(0000) knlGS:0000000000000000
nov 19 00:28:46 ratinkreesls kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
nov 19 00:28:46 ratinkreesls kernel: CR2: 000056198a565698 CR3: 000000011b21e000 CR4: 00000000003506e0
nov 19 00:28:46 ratinkreesls kernel: Call Trace:
nov 19 00:28:46 ratinkreesls kernel:  <TASK>
nov 19 00:28:46 ratinkreesls kernel:  ? rtw_restructure_ht_ie+0x4e2/0x510 [8192eu]
nov 19 00:28:46 ratinkreesls kernel:  ? __warn+0x81/0x130
nov 19 00:28:46 ratinkreesls kernel:  ? rtw_restructure_ht_ie+0x4e2/0x510 [8192eu]
nov 19 00:28:46 ratinkreesls kernel:  ? report_bug+0x191/0x1c0
nov 19 00:28:46 ratinkreesls kernel:  ? __x86_return_thunk+0x9/0x10
nov 19 00:28:46 ratinkreesls kernel:  ? prb_read_valid+0x1b/0x30
nov 19 00:28:46 ratinkreesls kernel:  ? handle_bug+0x41/0x70
nov 19 00:28:46 ratinkreesls kernel:  ? exc_invalid_op+0x17/0x70
nov 19 00:28:46 ratinkreesls kernel:  ? asm_exc_invalid_op+0x1a/0x20
nov 19 00:28:46 ratinkreesls kernel:  ? rtw_restructure_ht_ie+0x4e2/0x510 [8192eu]
nov 19 00:28:46 ratinkreesls kernel:  ? rtw_restructure_ht_ie+0x4e2/0x510 [8192eu]
nov 19 00:28:46 ratinkreesls kernel:  rtw_joinbss_cmd+0x3de/0x420 [8192eu]
nov 19 00:28:46 ratinkreesls kernel:  rtw_select_and_join_from_scanned_queue+0x62/0x1f0 [8192eu]
nov 19 00:28:46 ratinkreesls kernel:  rtw_do_join+0x10a/0x260 [8192eu]
nov 19 00:28:46 ratinkreesls kernel:  rtw_set_802_11_connect+0x11f/0x1c0 [8192eu]
nov 19 00:28:46 ratinkreesls kernel:  ? __x86_return_thunk+0x9/0x10
nov 19 00:28:46 ratinkreesls kernel:  cfg80211_rtw_connect+0x604/0xb70 [8192eu]
nov 19 00:28:46 ratinkreesls kernel:  ? ___slab_alloc+0x4f4/0x9c0
nov 19 00:28:46 ratinkreesls kernel:  cfg80211_connect+0x152/0x820 [cfg80211]
nov 19 00:28:46 ratinkreesls kernel:  ? __x86_return_thunk+0x9/0x10
nov 19 00:28:46 ratinkreesls kernel:  nl80211_connect+0x558/0x720 [cfg80211]
nov 19 00:28:46 ratinkreesls kernel:  genl_family_rcv_msg_doit.isra.0+0xe6/0x140
nov 19 00:28:46 ratinkreesls kernel:  genl_rcv_msg+0x1b6/0x2c0
nov 19 00:28:46 ratinkreesls kernel:  ? __pfx_nl80211_pre_doit+0x10/0x10 [cfg80211]
nov 19 00:28:46 ratinkreesls kernel:  ? __pfx_nl80211_connect+0x10/0x10 [cfg80211]
nov 19 00:28:46 ratinkreesls kernel:  ? __pfx_nl80211_post_doit+0x10/0x10 [cfg80211]
nov 19 00:28:46 ratinkreesls kernel:  ? __pfx_genl_rcv_msg+0x10/0x10
nov 19 00:28:46 ratinkreesls kernel:  netlink_rcv_skb+0x5b/0x110
nov 19 00:28:46 ratinkreesls kernel:  genl_rcv+0x28/0x40
nov 19 00:28:46 ratinkreesls kernel:  netlink_unicast+0x1a1/0x290
nov 19 00:28:46 ratinkreesls kernel:  netlink_sendmsg+0x254/0x4d0
nov 19 00:28:46 ratinkreesls kernel:  sock_sendmsg+0x96/0xa0
nov 19 00:28:46 ratinkreesls kernel:  ____sys_sendmsg+0x27d/0x310
nov 19 00:28:46 ratinkreesls kernel:  ? copy_msghdr_from_user+0x7d/0xc0
nov 19 00:28:46 ratinkreesls kernel:  ___sys_sendmsg+0x9a/0xe0
nov 19 00:28:46 ratinkreesls kernel:  ? __x86_return_thunk+0x9/0x10
nov 19 00:28:46 ratinkreesls kernel:  __sys_sendmsg+0x7a/0xd0
nov 19 00:28:46 ratinkreesls kernel:  do_syscall_64+0x5f/0xc0
nov 19 00:28:46 ratinkreesls kernel:  ? do_syscall_64+0x6b/0xc0
nov 19 00:28:46 ratinkreesls kernel:  ? __x86_return_thunk+0x9/0x10
nov 19 00:28:46 ratinkreesls kernel:  ? do_syscall_64+0x6b/0xc0
nov 19 00:28:46 ratinkreesls kernel:  ? do_syscall_64+0x6b/0xc0
nov 19 00:28:46 ratinkreesls kernel:  entry_SYSCALL_64_after_hwframe+0x77/0xe1
nov 19 00:28:46 ratinkreesls kernel: RIP: 0033:0x7f248d8d3910
nov 19 00:28:46 ratinkreesls kernel: Code: 00 f7 d8 64 89 02 48 c7 c0 ff ff ff ff eb b7 66 2e 0f 1f 84 00 00 00 00 00 90 80 3d d1 fc 0c 00 00 74 17 b8 2e 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 58 >
nov 19 00:28:46 ratinkreesls kernel: RSP: 002b:00007fff038eeb98 EFLAGS: 00000202 ORIG_RAX: 000000000000002e
nov 19 00:28:46 ratinkreesls kernel: RAX: ffffffffffffffda RBX: 000056198a4db980 RCX: 00007f248d8d3910
nov 19 00:28:46 ratinkreesls kernel: RDX: 0000000000000000 RSI: 00007fff038eebd0 RDI: 0000000000000006
nov 19 00:28:46 ratinkreesls kernel: RBP: 000056198a552080 R08: 0000000000000004 R09: 00007f248d99bd00
nov 19 00:28:46 ratinkreesls kernel: R10: 00007fff038eecb4 R11: 0000000000000202 R12: 000056198a4dbca0
nov 19 00:28:46 ratinkreesls kernel: R13: 00007fff038eebd0 R14: 0000000000000000 R15: 00007fff038eecb4
nov 19 00:28:46 ratinkreesls kernel:  </TASK>
nov 19 00:28:46 ratinkreesls kernel: ---[ end trace 0000000000000000 ]---
```

pbrochart, 2024-01-15:
Same error here:
memcpy: detected field-spanning write (size 16) of single field "ht_capie.mcs.rx_mask" at /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme.c:4744 (size 10)
Destination size (rx_mask) is smaller than the length required for the copy.

I replaced the line:
//memcpy(ht_capie.mcs.rx_mask, pmlmeext->default_supported_mcs_set, 16);
memcpy(&ht_capie.mcs, pmlmeext->default_supported_mcs_set, 16);

pbrochart, 2024-01-15:
More infos about this warning:
"Adjust memcpy() destination to be the named structure itself, rather than
the first member, allowing memcpy() to correctly reason about the size."
See https://lore.kernel.org/lkml/20210727205855.411487-10-keescook@chromium.org/

CGarces, 2024-01-16:
@pbrochart can you please send a PR?, I don't have free time to maintain this.

CGarces, 2024-01-20:
Fixed by #333

closed 2024-01-20 by CGarces