Mange/rtl8192eu-linux-driver #341: Does not compile under 6.8.1-arch1-1 (x86_64)

dafengsu7, 2024-05-18:
`DKMS make.log for 8192eu-r301.2741064 for kernel 6.9.1-arch1-1 (x86_64)
Sat May 18 11:22:16 PM CST 2024
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.9.1-arch1-1/build M=/var/lib/dkms/8192eu/r301.2741064/build  modules
make[1]: Entering directory '/usr/lib/modules/6.9.1-arch1-1/build'
  CC [M]  /var/lib/dkms/8192eu/r301.2741064/build/core/rtw_cmd.o
/var/lib/dkms/8192eu/r301.2741064/build/core/rtw_cmd.c:2069:6: warning: no previous prototype for ‘free_assoc_resources_hdl’ [-Wmissing-prototypes]
 2069 | void free_assoc_resources_hdl(_adapter *padapter, u8 lock_scanned_queue)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/8192eu/r301.2741064/build/core/rtw_cmd.c:2232:4: warning: no previous prototype for ‘_rtw_set_chplan_cmd’ [-Wmissing-prototypes]
 2232 | u8 _rtw_set_chplan_cmd(_adapter *adapter, int flags, u8 chplan, const struct country_chplan *country_ent, u8 swconfig)
      |    ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/8192eu/r301.2741064/build/core/rtw_cmd.c:2504:4: warning: no previous prototype for ‘rtw_ssmps_wk_hdl’ [-Wmissing-prototypes]
 2504 | u8 rtw_ssmps_wk_hdl(_adapter *adapter, struct ssmps_cmd_parm *ssmp_param)
      |    ^~~~~~~~~~~~~~~~
/var/lib/dkms/8192eu/r301.2741064/build/core/rtw_cmd.c:3295:6: warning: no previous prototype for ‘rtw_dynamic_chk_wk_hdl’ [-Wmissing-prototypes]
 3295 | void rtw_dynamic_chk_wk_hdl(_adapter *padapter)
      |      ^~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/8192eu/r301.2741064/build/core/rtw_cmd.c:3337:6: warning: no previous prototype for ‘lps_ctrl_wk_hdl’ [-Wmissing-prototypes]
 3337 | void lps_ctrl_wk_hdl(_adapter *padapter, u8 lps_ctrl_type, u8 *buf)
      |      ^~~~~~~~~~~~~~~
/var/lib/dkms/8192eu/r301.2741064/build/core/rtw_cmd.c:3500:6: warning: no previous prototype for ‘rtw_dm_in_lps_hdl’ [-Wmissing-prototypes]
 3500 | void rtw_dm_in_lps_hdl(_adapter *padapter)
      |      ^~~~~~~~~~~~~~~~~
/var/lib/dkms/8192eu/r301.2741064/build/core/rtw_cmd.c:3541:6: warning: no previous prototype for ‘rtw_lps_change_dtim_hdl’ [-Wmissing-prototypes]
 3541 | void rtw_lps_change_dtim_hdl(_adapter *padapter, u8 dtim)
      |      ^~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/8192eu/r301.2741064/build/core/rtw_cmd.c:3720:6: warning: no previous prototype for ‘rtw_dm_ra_mask_hdl’ [-Wmissing-prototypes]
 3720 | void rtw_dm_ra_mask_hdl(_adapter *padapter, struct sta_info *psta)
      |      ^~~~~~~~~~~~~~~~~~
/var/lib/dkms/8192eu/r301.2741064/build/core/rtw_cmd.c:3762:6: warning: no previous prototype for ‘power_saving_wk_hdl’ [-Wmissing-prototypes]
 3762 | void power_saving_wk_hdl(_adapter *padapter)
      |      ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/8192eu/r301.2741064/build/core/rtw_cmd.c:3768:6: warning: no previous prototype for ‘reset_securitypriv_hdl’ [-Wmissing-prototypes]
 3768 | void reset_securitypriv_hdl(_adapter *padapter)
      |      ^~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/8192eu/r301.2741064/build/core/rtw_cmd.c:4585:6: warning: no previous prototype for ‘btinfo_evt_dump’ [-Wmissing-prototypes]
 4585 | void btinfo_evt_dump(void *sel, void *buf)
      |      ^~~~~~~~~~~~~~~
/var/lib/dkms/8192eu/r301.2741064/build/core/rtw_cmd.c:5016:4: warning: no previous prototype for ‘rtw_c2h_wk_cmd’ [-Wmissing-prototypes]
 5016 | u8 rtw_c2h_wk_cmd(PADAPTER padapter, u8 *pbuf, u16 length, u8 type)
      |    ^~~~~~~~~~~~~~
/var/lib/dkms/8192eu/r301.2741064/build/core/rtw_cmd.c:5134:4: warning: no previous prototype for ‘session_tracker_cmd’ [-Wmissing-prototypes]
 5134 | u8 session_tracker_cmd(_adapter *adapter, u8 cmd, struct sta_info *sta, u8 *local_naddr, u8 *local_port, u8 *remote_naddr, u8 *remote_port)
      |    ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/8192eu/r301.2741064/build/core/rtw_cmd.c:5200:6: warning: no previous prototype for ‘session_tracker_chk_for_sta’ [-Wmissing-prototypes]
 5200 | void session_tracker_chk_for_sta(_adapter *adapter, struct sta_info *sta)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/8192eu/r301.2741064/build/core/rtw_cmd.c:5282:6: warning: no previous prototype for ‘session_tracker_chk_for_adapter’ [-Wmissing-prototypes]
 5282 | void session_tracker_chk_for_adapter(_adapter *adapter)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/8192eu/r301.2741064/build/core/rtw_cmd.c:5314:6: warning: no previous prototype for ‘session_tracker_cmd_hdl’ [-Wmissing-prototypes]
 5314 | void session_tracker_cmd_hdl(_adapter *adapter, struct st_cmd_parm *parm)
      |      ^~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/8192eu/r301.2741064/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/8192eu/r301.2741064/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/8192eu/r301.2741064/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/8192eu/r301.2741064/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/8192eu/r301.2741064/build/core/rtw_ieee80211.o
/var/lib/dkms/8192eu/r301.2741064/build/core/rtw_ieee80211.c:671:5: warning: no previous prototype for ‘rtw_get_akm_suite_bitmap’ [-Wmissing-prototypes]
  671 | u32 rtw_get_akm_suite_bitmap(u8 *s)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/8192eu/r301.2741064/build/core/rtw_ieee80211.c:1643:6: warning: no previous prototype for ‘dump_ht_cap_ie’ [-Wmissing-prototypes]
 1643 | void dump_ht_cap_ie(void *sel, const u8 *ie, u32 ie_len)
      |      ^~~~~~~~~~~~~~
/var/lib/dkms/8192eu/r301.2741064/build/core/rtw_ieee80211.c:1662:6: warning: no previous prototype for ‘dump_ht_op_ie_content’ [-Wmissing-prototypes]
 1662 | void dump_ht_op_ie_content(void *sel, const u8 *buf, u32 buf_len)
      |      ^~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/8192eu/r301.2741064/build/core/rtw_ieee80211.c:1676:6: warning: no previous prototype for ‘dump_ht_op_ie’ [-Wmissing-prototypes]
 1676 | void dump_ht_op_ie(void *sel, const u8 *ie, u32 ie_len)
      |      ^~~~~~~~~~~~~
[... 532 lines cut ...]
 2090 | int cfg80211_rtw_set_default_mgmt_key(struct wiphy *wiphy,
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/8192eu/r301.2741064/build/os_dep/linux/ioctl_cfg80211.c:2614:5: warning: no previous prototype for ‘rtw_cfg80211_is_target_wps_scan’ [-Wmissing-prototypes]
 2614 | int rtw_cfg80211_is_target_wps_scan(struct cfg80211_scan_request *scan_req, struct cfg80211_ssid *target_ssid)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/8192eu/r301.2741064/build/os_dep/linux/ioctl_cfg80211.c: In function ‘cfg80211_rtw_connect’:
/var/lib/dkms/8192eu/r301.2741064/build/os_dep/linux/ioctl_cfg80211.c:3899:32: warning: implicit conversion from ‘enum mlme_auth_type’ to ‘enum nl80211_auth_type’ [-Wenum-conversion]
 3899 |                 sme->auth_type = MLME_AUTHTYPE_SAE;
      |                                ^
/var/lib/dkms/8192eu/r301.2741064/build/os_dep/linux/ioctl_cfg80211.c: At top level:
/var/lib/dkms/8192eu/r301.2741064/build/os_dep/linux/ioctl_cfg80211.c:5227:6: warning: no previous prototype for ‘dump_station_parameters’ [-Wmissing-prototypes]
 5227 | void dump_station_parameters(void *sel, struct wiphy *wiphy, const struct station_parameters *params)
      |      ^~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/8192eu/r301.2741064/build/os_dep/linux/ioctl_cfg80211.c:5756:18: warning: no previous prototype for ‘rtw_sta_info_get_by_idx’ [-Wmissing-prototypes]
 5756 | struct sta_info *rtw_sta_info_get_by_idx(struct sta_priv *pstapriv, const int idx, u8 *asoc_list_num)
      |                  ^~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/8192eu/r301.2741064/build/os_dep/linux/ioctl_cfg80211.c:9792:5: warning: no previous prototype for ‘cfg80211_rtw_external_auth’ [-Wmissing-prototypes]
 9792 | int cfg80211_rtw_external_auth(struct wiphy *wiphy, struct net_device *dev,
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
make[3]: *** [scripts/Makefile.build:244: /var/lib/dkms/8192eu/r301.2741064/build/os_dep/linux/ioctl_cfg80211.o] Error 1
make[2]: *** [/usr/lib/modules/6.9.1-arch1-1/build/Makefile:1919: /var/lib/dkms/8192eu/r301.2741064/build] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.9.1-arch1-1/build'
make: *** [Makefile:2287: modules] Error 2
`

CGarces, 2024-05-18:
Please confirm that works after #342

closed 2024-05-18 by CGarces

dafengsu7, 2024-05-21:
Everything is ok, Thank you a lot!