Mange/rtl8192eu-linux-driver #349: Does not compile under kernel 6.13.0-1-mainline

Gen5551, 2025-01-21:
|             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
os_dep/linux/ioctl_cfg80211.c:279:20: предупреждение: «nl80211_chan_width_str» определена, но не используется [-Wunused-function]
  279 | static const char *nl80211_chan_width_str(enum nl80211_chan_width cwidth)
      |                    ^~~~~~~~~~~~~~~~~~~~~~
os_dep/linux/ioctl_cfg80211.c:234:13: предупреждение: «rtw_get_chbw_from_nl80211_channel_type» определена, но не используется [-Wunused-function]
  234 | static void rtw_get_chbw_from_nl80211_channel_type(struct ieee80211_channel *chan, enum nl80211_channel_type ctype, u8 *ht, u8 *ch, u8 *bw, u8 *offset)
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
os_dep/linux/ioctl_cfg80211.c:217:34: предупреждение: «rtw_chbw_to_nl80211_channel_type» определена, но не используется [-Wunused-function]
  217 | static enum nl80211_channel_type rtw_chbw_to_nl80211_channel_type(u8 ch, u8 bw, u8 offset, u8 ht)
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
os_dep/linux/ioctl_cfg80211.c:201:20: предупреждение: «nl80211_channel_type_str» определена, но не используется [-Wunused-function]
  201 | static const char *nl80211_channel_type_str(enum nl80211_channel_type ctype)
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~
make[4]: *** [/usr/lib/modules/6.13.0-1-mainline/build/scripts/Makefile.build:194: os_dep/linux/ioctl_cfg80211.o] Ошибка 1
make[3]: *** [/usr/lib/modules/6.13.0-1-mainline/build/Makefile:1989: .] Ошибка 2
make[2]: *** [/usr/lib/modules/6.13.0-1-mainline/build/Makefile:251: __sub-make] Ошибка 2
make[2]: выход из каталога «/var/lib/dkms/rtl8192eu/1.0/build»
make[1]: *** [Makefile:251: __sub-make] Ошибка 2
make[1]: выход из каталога «/usr/lib/modules/6.13.0-1-mainline/build»
make: *** [Makefile:2287: modules] Ошибка 2
`

Автор, подготовьте пожалуйста патчи под 6.13!!!
Полный файл make.log

[make.log](https://github.com/user-attachments/files/18487790/make.log)

GenocideStomper, 2025-02-10:
Having this issue with 6.13.2-zen while installing through the AUR:

```
==> dkms install --no-depmod rtl8192eu/5.6.4.r302.g2fddecc -k 6.13.2-zen1-1-zen

Error! Bad return status for module build on kernel: 6.13.2-zen1-1-zen (x86_64)
Consult /var/lib/dkms/rtl8192eu/5.6.4.r302.g2fddecc/build/make.log for more information.
==> WARNING: `dkms install --no-depmod rtl8192eu/5.6.4.r302.g2fddecc -k 6.13.2-zen1-1-zen' exited 10
```

This is the end of the mentioned make.log:

```
os_dep/linux/ioctl_cfg80211.c:9953:32: note: (near initialization for ‘rtw_cfg80211_ops.set_monitor_channel’)
make[4]: *** [/usr/lib/modules/6.13.2-zen1-1-zen/build/scripts/Makefile.build:196: os_dep/linux/ioctl_cfg80211.o] Error 1
make[3]: *** [/usr/lib/modules/6.13.2-zen1-1-zen/build/Makefile:1987: .] Error 2
make[2]: *** [/usr/lib/modules/6.13.2-zen1-1-zen/build/Makefile:251: __sub-make] Error 2
make[2]: Leaving directory '/var/lib/dkms/rtl8192eu/5.6.4.r302.g2fddecc/build'
make[1]: *** [Makefile:251: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.13.2-zen1-1-zen/build'
make: *** [Makefile:2287: modules] Error 2

# exit code: 2
```

etzngr, 2025-02-17:
https://github.com/Mange/rtl8192eu-linux-driver/pull/345 should fix this. @CGarces could you please merge this one?

CGarces, 2025-02-23:
Fixed

closed 2025-02-23 by CGarces

Gen5551, 2025-02-27:
Thanks you! Wifi working from kernel 6.14 rc4 mainline!